analogdevicesinc/TransceiverToolbox #31: modem_qpsk

khan2018, 2020-07-22:
Hi

The property Upsampling factor is not defined in the QPSKChannel.m and is being used in the main file EndtoEndTest.m

Also, there is an error regarding the delay
Displaying stack trace:
Error using dsp.VariableFractionalDelay>VariableFractionalDelay.setParameters
	   • In dsp.VariableFractionalDelay>VariableFractionalDelay.setParameters
	   • In QPSKChannel>QPSKChannel.stepImpl (line 83)
	   • In EndToEndTest (line 209)

Thanks,

tfcollins, 2020-07-23:
Looks like QPSKChannel got updated a bit. I'll have a fix later today. Thanks for reporting.

khan2018, 2020-07-27:
I still am having the same errors in the script I wonder if the script is updated or I didn't download the updated script.

Thanks

tfcollins, 2020-07-27:
The code has not been updated yet. There is a task in the queue but there are a few things in front right now.

-Travis