analogdevicesinc/TransceiverToolbox #34: Modem-QPSK Transmitter Compilation Error in MATLAB R2020a

sunipkmukherjee, 2020-07-27:
The transmitter HDL fails to compile with an error where uint64 is written into embedded.fi {uint64} type, an issue not present in MATLAB R2019b.

Diagnostic:
This assignment writes a 'uint64' value into a 'embedded.fi {uint64}' type. Code generation does not support changing types through assignment. Check preceding assignments or input type specifications for type mismatches.

Function 'Transmitter HDL/DMA Interface/DMA Control' (#127.2966.2976), line 90, column 13:
"wordBuffer"

Fix: Initialize the wordBuffer variable with a fixdt(0,64,0,0) so that there is no type mismatch further down in the code.

tfcollins, 2020-07-27:
Thanks for reporting. We are still in the process of updating everything to R2020a.

-Travis

sunipkmukherjee, 2020-07-28:
Should I make the change in the demos and submit a pull request?