analogdevicesinc/TransceiverToolbox #78: Problem with the Fast Frequency Hopping Example

sdrlab, 2021-06-27:
we have a question about Frequency Hopping Example of the RF SOM with Transceiver Toolbox, we followed the tutorial and finished all the steps include building kernel, and when we run the fastlock.m script in matlab we get an error says "Abstract classes cannot be instantiated.  Class 'FrequencyHopper' is declared as Abstract.' , we are using matlab 2020b and vivado 2018.3 as said in the requirements, we also installed libio as said in the starting guide, and the required compiiler.

tfcollins, 2021-06-27:
Which release of the toolbox are you using?

sdrlab, 2021-06-28:
We are using version 20.2.1
[image]

tfcollins, 2021-06-28:
This was fixed here: https://github.com/analogdevicesinc/TransceiverToolbox/commit/90db749365b73dd50d67da07aecd600bb93d985c

I've update the MLTBX file on the github release page. Either grab the FrequencyHopper.m file directly or re-install the toolbox.

closed 2021-06-28 by tfcollins