analogdevicesinc/TransceiverToolbox #82: ADRV9002 Interface Mapping for Valid signals

UsernameUsurper, 2021-08-11:
There seems to be an error in the VALID signal interface map for ADRV9002 targeting, and the naming convention seems confusing.

There are only 3 valid signals (with directions):
'IP Data Valid OUT' (out) - Assuming this is the data out of the IP to the transceiver.
'IP Data Valid OUT' (in) - Same name as above. Assuming this is the baseband data from the transceiver's ADC.
'IP Valid Tx Data IN' (in) - Assuming this is the data into the IP.

So, we are then just missing the baseband TX valid out?

There is also a commented out 'IP Load Tx Data OUT' (out).

[add_io.txt](https://github.com/analogdevicesinc/TransceiverToolbox/files/6966192/add_io.txt)

tfcollins, 2021-08-11:
The TX side does not use any output valid signals since the interface core does not have any input valids for the DAC. This is logical since the DAC always expects samples.

UsernameUsurper, 2021-08-11:
Thanks Travis, thought the interfaces would be the same as the FMComm examples.

closed 2021-08-11 by UsernameUsurper

tfcollins, 2021-08-11:
FMComms have FIFOs so you need the valid signals there.