analogdevicesinc/TransceiverToolbox #130: Library path causes ZCU102+ADRV9002 HDL Coder project build to fail on R2021b

m4eme, 2022-11-17:
On both Windows and Linux, using HDL Coder to target the ZCU102+ADRV9002 combo, the 21.2.1 release fails while building IPs in the HDL Coder 'Create Project' phase.

Vivado looks for the library folder in either C:/library (Windows) or $HOME/library (Linux) instead of in the copy made in the project folder. 

I have tried with both the mltbx file as well as with my own build. The failure mode is the same.

Thanks
--
Mariano Maté

```
Starting Transceiver Toolbox HDL build
Building:
- axi_adrv9001
- axi_dmac
- axi_sysid
- sysid_rom
- util_pack/util_cpack2
- util_pack/util_upack2
Please wait, this might take a few minutes
ERROR: Wrong path to IP or the IP does not have a Makefile starting from "C:/library/axi_adrv9001" 
```

tfcollins, 2022-11-17:
Can you export and post your configuration from HDL Workflow Advisor

tfcollins, 2022-11-18:
Things are working fine with the release on my end. Can you provide the full log file generated from step 4.1

m4eme, 2022-11-21:
Thanks Travis, here is the log.
[workflow_task_CreateProject.log](https://github.com/analogdevicesinc/TransceiverToolbox/files/10055288/workflow_task_CreateProject.log)

m4eme, 2022-11-21:
I found the issue after some debugging.

The projects I work on are stored under C:\projects, so my working directory is e.g.:
```C:\projects\P12736\work\src\```

Now line 65 in hdl/vendor/AnalogDevices/vivado/projects/scripts/adi_make.tcl uses the following to retrieve the root_hdl_folder
``` regsub {/projects.*$} $glb_path "" root_hdl_folder ```

This deletes all characters in path after the first match to ```projects```. In my case it returns simply ```C:```

I've worked around the issue with this clumsy Tcl replacement for adi_make.tcl:65
```set root_hdl_folder [string range $glb_path 0 [expr [string last projects $glb_path]-2]]```

tfcollins, 2022-12-01:
Sorry for the delay. Have been out on travel.

A question here, does your machine have a particular locale?

m4eme, 2023-01-16:
Sorry for the very late reply. I have workaround this issue by pointing HDL Coder to a folder that does not contain 'projects' in the path.

My locale was es_ES.UTF-8 back when I opened the issue. Now I've tested this on a machine with locale en_US.UTF-8 and the issue persists...

tfcollins, 2023-01-26:
We tried this on several different machines and cannot repeat the issue.

Probably the next setup is to verify you can manually build the HDL. This would be done outside of MATLAB in a terminal like so:

```bash
<source Vivado 2019.1 tools>
git clone -b hdl_2019_r2  https://github.com/analogdevicesinc/hdl.git
cd hdl/projects/adrv9001/zcu102
vivado -mode tcl
source ../../scripts/adi_make.tcl
adi_make::lib all
source ./system_project.tcl
```

This is essentially what MATLAB does.